DukeRobotics/robosub-ros #83: Write Task superclass

davidAmiron, 2019-10-19:
Write the class task that will be overrided by all other tasks

neildhar, 2020-01-20:
I'd like to propose a change to the task architecture. There are a couple issues I identified with the current one:

- Difficulty of creating subtasks
- Constantly creating and destroying subscribers and publishers for every task is confusing
- We need simple, easy to construct tasks to make the program easy to reason about

The current task architecture is pretty complicated. Every task makes its own subscriber and publisher (which might even fail if the previous publisher was not correctly destroyed). They also store a lot of state and generally share a lot of functionality.  I think there are a couple of changes we should make:

## Separate utility functions
 Pull out the utility functions (such as distance calculation, moving, angles, and pose transforms) into a separate module called TaskUtils. These functions would take at most an additional Pose parameter and they would be easier to generalise and reason about.

## Consolidate shared state
Somehow make the state that is shared across tasks global (this includes the current location and the handles for the publisher and subscriber). This could mean that the top level module (currently task_planner.py) stores all this state and passes a handle to each task when it is initialised. Ideally, we'd just make these static members of TaskPlanner but I don't know if you can do something like that in Python.

## Change TaskBase
With these changes, the tasks should become very lightweight, with almost no state (apart from pose at task start) and only the `init` and `run` methods. The `init` function in TaskBase should now only need to initialise what little state there is and would almost never need overriding.

## Implement template tasks

This is probably the most interesting part. We create a series of derived classes that just need a constructor argument at instantiation to become a usable task. Off the top of my head, I can think of the following tasks:

- **MoveToGlobal:** Given desired global pose as a constructor argument (relative to mission start), end the task at that pose.
- **MoveToLocal:** Same but with a local pose relative to task start
- **TimedMove:** Move in a given direction for a given amount of time
- **FollowState:** Given a callback function or ROS topic corresponding to some processed sensor data (say, the target position/velocity desired to hit a buoy), set the robot's desired state to correspond to that. This is useful for generalising to a lot of tasks like the pinger, buoy, etc and provides a standardised interface for the people developing that code.
- **ListTask:** Given a list of tasks, initialise and execute each one in order. This is useful for composing tasks and creating subtasks. Having these will make it easier to organise tasks into logical units and avoid making tasks too big.

With this organisation, it will be possible to execute almost the entire mission without needing to write much code. Each mission objective (e.g. move through gate, go to buoy) could just be instantiated as a ListTask made up of one or more MoveTo* tasks and FollowState tasks. These mission objectives could then in turn be wrapped into a top level ListTask. The top level entity that replaces task_planner would therefore only need to be able to launch and run a single task in order to execute the entire mission.

davidAmiron, 2020-01-22:
Just thinking about this, probably will need to create the publisher and subscribers in the task runner and pass them in, static variables exist in python but these will need to be created at runtime which I'm not sure happens with static variables.

davidAmiron, 2020-01-22:
What you propose with the tasks inheriting from others we did already plan on doing, as well as the ListTask (or its equivalent). We will have one main Task that will have everything inside it, and then the task runner will just run that.

davidAmiron, 2020-01-22:
I do like the idea of the task utils file, there is a lot of annoying ROS stuff that we can wrap up in that, like transformations.

neildhar, 2020-01-22:
> Just thinking about this, probably will need to create the publisher and subscribers in the task runner and pass them in, static variables exist in python but these will need to be created at runtime which I'm not sure happens with static variables.

You can create them statically and then assign them at runtime in the constructor for task planner.

closed 2020-01-25 by davidAmiron