DukeRobotics/robosub-ros #105: Scale output from thruster controls properly

ShaanGondalia, 2019-11-23:
Current Algorithm:
- val = max(abs(allocations))
- If val>1, allocations=allocations/val

This doesn't give us the correct output for thruster allocations. For example, moving straight forward with a max value from PID should result in max allocations from 4 of the thrusters (1, 1, -1, -1), but currently our allocations look something like (0.3, 0.3, -0.3, -0.3).

5happy1, 2020-01-11:
we should ask Zavlanos for help on why the outputs are scaled the way they are

rabinowitz, 2020-09-17:
We got help from Professor Leila Bridgeman over the summer and confirmed that the way we are currently scaling is probably the best we're going to get for this version of the controls code.

closed 2020-09-17 by rabinowitz