DukeRobotics/robosub-ros #124: fix dyld error for macs

ncitron, 2020-01-11:
running `python rosvrepcomm.py` on mac produces:
`dyld: warning, LC_RPATH @executable_path/Frameworks in /Users/noahcitron/Developer/ros-simulate/remoteApi.dylib being ignored in restricted program because of @executable_path
Program started`

puzzler7, 2020-01-11:
Everything I see points to either disabling SIP, or editing/making the dylib ourselves. Here's some of the links I've been looking through.:

=======

Issue should be fixed by disabling System Integrity Protection. However, this requires restarting the computer in recovery mode, which is onerous and undesirable.
https://stackoverflow.com/questions/38534364/tensorflow-mac-gpu-pywrap-tensorflow-ignored-in-restricted-program

Remove suid flag (about halfway down)?
https://theevilbit.github.io/posts/dyld_insert_libraries_dylib_injection_in_macos_osx_deep_dive/

Maybe we need to get xcode and change/edit the library ourselves?

CoppeliaSim suggests we might need to build the libraries ourselves:
http://www.coppeliarobotics.com/helpFiles/en/remoteApiClientSide.htm

Can we disable Library Validation Entitlement?
https://developer.apple.com/documentation/bundleresources/entitlements/com_apple_security_cs_disable-library-validation

puzzler7, 2020-01-12:
Just need to use python3 to run the thing, as python3 isn't a restricted program.

closed 2020-01-18 by puzzler7