DukeRobotics/robosub-ros #288: Automate startup of roscore with systemd

muthuArivoli, 2021-06-05:
## Background
Ideally, we would like the landside container to be the entrypoint of any human interaction with the stack. The GUI is planned to handle things like launching launch files and nodes and to call services and the like by creating an ssh connection and doing so, but the issue still remains that to launch the GUI, you need roscore already running (so the GUI can't launch roscore itself, since it needed it to run before running). Thus, we should automate this startup at boot of the container, using systemd to make it easy to shutdown/restart when necessary.

## Task
Use this (https://blog.roverrobotics.com/how-to-run-ros-on-startup-bootup/) as a very general guide. Note that it should be heavily modified (only need the roscore.service file, and it should not be forking). Add it to the docker image.

ShaanGondalia, 2022-09-04:
Will begin work on this once GUI is debugged and merged.

muthuArivoli, 2022-09-04:
Ok, so I did some preliminary investigation into this last year, and I realized that it's not possible to run systemd in a container with a Windows or Mac host, like all of our dev computers, due to the way processes are handled (docker delegates that to the host kernel I think). It may be possible to use supervisord see [this](https://docs.docker.com/config/containers/multi-service_container/), but not completely sure what that is or how it works.

ShaanGondalia, 2022-09-05:
Is it not sufficient to start roscore via `entrypoint.sh` on onboard, similarly to starting ssh-server in [core](https://github.com/DukeRobotics/robosub-ros/blob/master/docker/core/entrypoint.sh)? My understanding is that entrypoint is run whenever the container is started, so we can use `docker stop` and `docker start` to shutdown/restart roscore.

muthuArivoli, 2022-09-05:
I'm not completely sure on this, but I think entrypoint is only run when the container is created, not when it's started after being stopped (but I could wrong about that). If that's the case, then to avoid all the network messiness the docker-compose file takes care of, I think you would have to do a `docker-compose down` and then `up`, and that sounds somewhat tedious, I haven't run stuff in a while but I think it still takes a while to do that.

ShaanGondalia, 2022-09-05:
I see, that makes sense. I'll do some testing (with the new build command 😄) to see what the behavior is.

AndrewCouncil, 2022-09-10:
I think potentially that adding a `command` field to the docker compose file where the command specifies to start roscore (potentially in the background) could also work. From my experience, I think this command should run every time the container is stopped and started again. I will try testing this out.

ShaanGondalia, 2022-09-10:
My understanding is that `command` overwrites CMD in the Dockerfile, although I'm not sure how that interacts with Entrypoint. My assumption is that the command wont run because entrypoint will be prioritized but give it a shot

AndrewCouncil, 2022-09-24:
Order of operations for initial startup of onboard:
1. Start SSH server
2. Run build.sh script
3. Source setup.bash
4. Start roscore
5. Run bash

Order of operations for initial startup of landside:
1. Start SSH server
2. Run bash

Currently thinking build only needs to run when container is executed with "docker run" or "docker compose up" (only on initial startup). It is rare that we stop a container and "docker start" that container again, usually we just "docker run" the image again, and now we will run build.sh on an image run and will *not* on container start. This matches well with a common docker workflow and the way we start up our containers.

vedarshshah, 2023-08-31:
Automating these operations is not a priority for now, although we may consider it in the future.

closed 2023-08-31 by vedarshshah