DukeRobotics/robosub-ros #294: Migrate task planning to use SMACH

muthuArivoli, 2021-06-05:
## Background
Task planning is complex, rather that creating our own library and maintaining it (which we tried in v2, but there are many limitations), we can just abstract it to use an already existing library. SMACH appears to be a simple library, written in Python, that abstracts tasks to state machines, and allows easy composition of tasks.

## Task
We should still maintain a similar inheritance structure, with the Task class being the superclass of all tasks, but it should use SMACH internally to run/maintain the execution style of task planning. Design will need to be thoroughly reviewed to ensure that it is easily extensible.

azhou202, 2021-06-12:
- [ ] delete useless metatasks @azhou202 
- [ ] implement style task smach @AndrewCouncil 
- [ ] make new branch off of master and add task-planning-gate useful code
- [ ] test smach in simulation 
- [ ] update documentation with smach
- [ ] get simultaneous tasks working

ShaanGondalia, 2022-09-03:
Closing, stale issue.

closed 2022-09-03 by ShaanGondalia