DukeRobotics/robosub-ros #303: Add models of competition objects to simulation

muthuArivoli, 2021-09-15:
Need to add all the models of objects from competition to simulation. At the end, ideally, we should have a scene representing each task (with only the necessary objects in it for the task), and a overall big scene representing the competition (with all objects and scenes), to make it easy for task planning to test.

List of objects to be modeled:
TODO

samuel-thompsonn, 2021-09-25:
On the [add-sim-models branch](https://github.com/DukeRobotics/robosub-ros/tree/add-sim-models/landside/catkin_ws/src/simulation/personal/models) we have some basic rectangular prisms with the same dimensions as the contest props for the following:
- Gate
- Image of each type, from the ['Choose Your Side'](https://robonationforum.vbulletin.net/forum/robosub/robosub-2021/3555-choose-your-side-gate) task
- Buoy of each type, from the ['Make the Grade'](https://robonationforum.vbulletin.net/forum/robosub/robosub-2021/3558-make-the-grade-buoys) task
- Bin without lid
- Octagon (Octagon.ttm is a box, and Octagon_2.ttm is an actual octagon)
- Straight path marker
- Torpedo

vedarshshah, 2023-08-31:
This issue was written for CoppeliaSim, which we have decided to stop using. We will create new models for the latest competition props for our new sim.

closed 2023-08-31 by vedarshshah