DukeRobotics/robosub-ros #351: Replace desired state topics with actions

ShaanGondalia, 2022-09-29:
## Background
Using topics for desired state in controls is an anti-pattern. Currently, the `desired_state` node expects constant publishing of a set-point (at a rate of 10Hz) to run the PID loops and produce movement. If the node doesn't receive an update on any desired state topic, it halts all movement until it receives a new message. It also halts if more than one type of control is requested in the same cycle.

This design better matches the use case of a [ROS Action](https://wiki.ros.org/actionlib), as we can model any control as an action that is requested of the controls package. The action specification includes a goal, feedback, and result. Because controls is stateless (it doesn't keep track of when a movement is complete), the feedback and result portions of the action seem useless and can be discarded. The goal can be the desired state. 

While controls doesn't need the power of feedback and result, switching to actions eliminates the need for constant publishing that can cause issues if timings are mis-aligned and simplifies logic in `desired_state.py`. Also, this should improve the process of calling controls from task planning; action calls preempt previous actions by default.

## Important Notes
Here are two designs for switching to actions:

1. Write 3 different actions for pose, velocity, and power control. This requires the creation of 3 action servers in `desired_state.py`. This is a problem because it allows for simultaneous actions (i.e. a request for velocity control wont automatically cancel a pose control action that's already running). This simplifies the calling conventions but doesn't take advantage of the built in preemption in actionlib.
1. Write 1 action that encompasses pose, velocity, and power control. The goal of this action contains the desired set-points as well as the type of control requested. This complicates the calling conventions but imposes the constraint that only one request is acted upon at any given time.

## Task
- [ ] Write custom messages and actions to replace desired state topics
- [ ] Write action servers in desired state
- [ ] Update task planning scripts and `test_state_publisher.py`
- [ ] Update pub/sub documentation in Controls README (Maybe include boilerplate for creating an action-client?)

Brokemia, 2022-11-05:
Part of the reason for using actions is to take advantage of the preempt feature, not just as mentioned in the original issue, but also to stop existing movement (for example when task planning detects the robot has reached its desired pose.

Brokemia, 2022-11-12:
We've made some plans to refactor our task planning structure. I don't want to delay this issue to work on another issue (since this is already a side-track from gate task), so the plan is to make a minimal version of this, merge it, and then work on the larger refactor.

This minimal version will fix `test_state_publisher`, do the documentation, and then make some really quick changes to task_planning (probably just shoving some clients into TaskState.

PR #383 (2022-12-03): Change controls to use ROS Actions
This changes controls so that instead of having to publish to specific topics at a fixed rate, we use actions to send "goals" which run until pre-empted by another goal or canceled manually. The relevant task-planning changes that come as part of this PR have been done hastily, due to our plans to refactor some of our task planning system next.

Closes #351
closed 2022-12-03 by Brokemia