DukeRobotics/robosub-ros #358: CV Sprint 2 Stereovision: Building Towards the CV Control Panel

vedarshshah, 2022-10-14:


ShaanGondalia, 2022-10-14:
This seems related to #311, which is a custom GUI that we plan to use for launching files and editing params on the fly. The code is almost ready to be merged, it's just waiting on the controls refactoring in #319 before getting merged. Maybe we can consider creating another rqt widget that will serve as this cv control panel?

vedarshshah, 2022-10-14:
I considered building a GUI in python for the CV control panel, but I think it makes more sense to make it a web interface using rosbridge and Foxglove due to a couple of advantages.
- Using Foxglove cuts down on development time for the GUI significantly.
  - All the heavy-lifting for getting data in/out of the websocket and showing image streams is already done. The only functionality we need to add is executing roslaunch and rosbag.
  - Any customizations/extensions made for Foxglove would be developed with React & JS. I'm sure members would be more familiar with React, JS, or web development in general than with rqt. And it would be easier to create the same interface using React/HTML/CSS than with rqt.
- The control panel can be accessed wirelessly from computer(s) other than the one plugged into the robot.
  - If the computer plugged into the robot launches rosbridge on landside, is connected to WiFi, and provides its IP address, other computers connected to the same WiFi can load Foxglove with this IP address and view the control panel.
  - Thus, if another subteam needs to use the computer plugged into the robot, CV can still access the CV control panel independently on another computer.
- The Foxglove GUI is way nicer than what we could develop with rqt (obviously not super important).

I can look into reusing code from #311 related to launching. But overall, I think it makes more sense to make the CV control panel a web interface instead of a rqt widget, and using Foxglove specifically makes things much easier.

ShaanGondalia, 2022-10-15:
> Using Foxglove cuts down on development time for the GUI significantly.

I disagree, writing a the qt widget logic is very straightforward, and will allow you to leverage existing ros packages for viewing images. I'm imaging buttons for a few launch files and a camera feed, which is trivial to implement using image_view (we already use this package for viewing mono camera feeds). I think @muthuArivoli knows more about automatically generating the ui files as he did that for #311.

> Any customizations/extensions made for Foxglove would be developed with React & JS.

This is a good point, definitely a useful and transferable skill for team members to learn.

> The control panel can be accessed wirelessly from computer(s) other than the one plugged into the robot.

This is already possible with our current setup. The onboard robot computer connects to a router via the tether, then users can connect wirelessly (or via ethernet) to the router's network and run landside containers to view camera feeds or ssh into the robot computer themselves.

> The Foxglove GUI is way nicer than what we could develop with rqt (obviously not super important).

Valid but perhaps overkill for our purposes. I don't have the full picture of the end goal of this Control Panel, what functionality do you expect it to have in the future? 

Also, I'm only presenting these thoughts to give you an additional perspective on this topic. If the team thinks FoxGlove (or any other solution) is better than a qt-based GUI and has interest then you should definitely go for it.

vedarshshah, 2022-10-15:
@ShaanGondalia, I really appreciate your perspective on this and thanks for correcting my assumptions. I have asked @ricohasgithub and @tylerfeldman321 to take a look at this issue and add their thoughts as well.

> I don't have the full picture of the end goal of this Control Panel, what functionality do we expect it to have in the future?

Functionality expected from the control panel
- View multiple image streams from selected rostopics simultaneously (at least 6), including raw image streams and image streams with bounding boxes overlaid on top.
- Execute roslaunches with the press of a button.
- Execute rosbags with the press of a button (the bagfile should be stored on landside).

You can take a look at [this file](https://drive.google.com/drive/u/0/folders/1pxMzNo0hkNRzczM6a6M7dMt9YE-xUzrt) on our Google Drive that provides a visualization of these requirements. The file is not meant to represent the final design of the panel, it's just an illustration of the requirements.

> This is a good point, definitely a useful and transferable skill [React & JS] for team members to learn.

Yes, but there wouldn't be much work to do for React & JS since the extension would be simple; at best 1-2 team members would get to do a short React project.

With your insight, I'm thinking using rqt would be fine and would keep it consistent with the other control panel. Maybe we could integrate both control panels into one?

ShaanGondalia, 2022-10-15:
Based on the ui mockup, `rqt` is well-served to handle everything that we would need. Buttons are trivial to implement. I'm not 100% sure on embedding camera feeds inside of a custom widget. The `rqt_image_view` package exists but we'd have to research that further. [This](https://answers.ros.org/question/200876/rqt-custom-imageview-showing-images-from-different-cameras/) might be a good place to start. If that doesn't work then we may have to go with an external solution like FoxGlove.

> Can you elaborate more on these limitations?

`rqt` plugins are written for qt, so they can't leverage powerful JS frameworks. However, we can use tools like [qt designer](https://doc.qt.io/qt-6/qtdesigner-manual.html) to automate the generation of the .ui files that we need, which can be easier than writing the front-end manually. Qt in general is probably a lot less powerful than most JS front-end frameworks but that isn't a huge deal for us.

> Maybe we could integrate both control panels into one?

The way I see it, if we go with `rqt` then there are two options:
1. Create an rqt plugin that is a standalone CV control panel.
2. Create an rqt widget that fits into the existing gui plugin. This can be thought of as another "tab" in the existing control panel.

I would probably recommend making a standalone plugin because it seems like we want to spin it up without the rest of the gui stuff, but I think either way works. As a side note, one future plan for the main gui was to add a camera widget that would show camera feeds, but we haven't gotten around to it yet.

vedarshshah, 2022-11-05:
Completed Sprint 2 as of October 29, 2022.

closed 2022-11-05 by vedarshshah