DukeRobotics/robosub-ros #371: Change how we use SMACH in task planning

Brokemia, 2022-11-05:
## Background
Currently, we've used SMACH very similarly to the way we used our old task framework. Our `Task` class has been changed to extend `smach.State` and we have all our tasks extend that `Task` class. SMACH allows for entire state machines to be used as states in other state machines. These state machines obviously can't derive from `Task`, as they already need to extend `smach.StateMachine`.

Our `Task` class currently does very little. It pretty much just exists to hold CV data and otherwise works very similarly to `smach.State` itself. We should probably look into removing the `Task` class and then making our tasks extend `smach.State` and `smach.StateMachine` directly.

## Tasks
- [ ] Remove `Task` class
- [ ] Move CV functionality somewhere else
- [ ] Convert existing tasks to directly extend `smach.State`

ShaanGondalia, 2022-11-12:
Notes from a discussion between @Brokemia and myself regarding our Task planning system:

We plan on refactoring our system into the following (naming conventions are subject to change):

1. `task_planner` is a ROS Node that serves as the brain of task planning. Contains a smach state machine that controls which **Obstacle Task** is running at any given time. Also creates **Interface** ROS Nodes that are passed down to obstacle tasks.
2. **Obstacle Tasks** are smach state machines responsible for reaching a high-level goal for an obstacle. Each state in this state machine is a **Basic State**. Potential examples include `gate_task`, `octagon_task`, `follow_path_markers_task`, etc. These pass the **Interface** ROS Nodes to the **Basic States** that compose the task.
3. **Basic States** are smach states that represent a simple, low-level goal. Examples include `move_to_local_pose`, `calculate_gate_location`, and `sweep_for_cv_obect`. These states are non-blocking by design.
4. **Interfaces** are ROS Nodes that contain APIs for accessing data from outside of Task Planning. We currently expect a few interfaces:
	* `controls_interface` exposes methods for retrieving the state of the robot and calling controls. Example methods include `get_state()`, `move_to_local_pose()`, `move_to_global_pose()`, `enable_controls()`, and `move_with_velocity()`.
	* `cv_interface` exposes methods for retrieving bounding box data from the cv package.

Some motivating questions for this refactoring are:

## Why are we doing this?
Task planning has gone through several re-designs over the past few years. This has involved retrofitting solutions onto pre-existing code without a clear direction on the overall design of our system. This refactoring aims to clearly define some design principles that we'll stick to, as well as clean up our code base for future members.

## Is there a difference between a state and state machine?
In smach, state machines are extensions of states. As a result, they can be used interchangeably. We chose to clearly define Obstacle Tasks as state machines because of our human intuition on completing tasks. Completing an obstacle involves many smaller fundamental tasks, which we represent as states. We feel that this distinction adds some clarity to our architecture.

## Why abstract ROS code to Interfaces?
1. With the existing system, we create a publisher and subscriber for every fundamental task that needs to access some data outside of task planning. We've gotten around this in the past by creating singletons, but the implementation of this is a little convoluted. If we abstract our ROS code to interfaces, we can keep the singleton design pattern without overly complicating our system.
2. Currently, there's no feedback that indicates that multiple tasks are publishing to controls at the same time. Using Interfaces allows task planning to have a centralized "publishing" hub that can keep track of potential issues while interfacing with controls or any other package.
3. ROS2 requires all ROS interfacing code (pub, sub, srv, action) to be declared explicitly in a Node. Extracting our code to Interface ROS Nodes now will make migration simpler in the future.

## Why stick with smach?
This is already a large refactoring so to keep the scope reasonable we won't change to another state machine package. We will only use the python functionality of smach, not its ROS interface. When we migrate to ROS2 we'll create a fork of the [python smach implementation](https://github.com/ros/executive_smach/tree/noetic-devel/smach/src/smach) and add it to our code-base. We have looked into some more popular python packages like [`transitions`](https://github.com/pytransitions/transitions), but figured that smach is best suited for us at the moment.

## Why should all Basic States be non-blocking?
Right now, some of our tasks are blocking in the sense that they loop ad-infinitum until they get the desired result. For instance, a `move_to_pose` task will loop until the state of the robot is within a certain threshold. The parent state machine then has to wait until this condition is met before moving on to another state. Instead of having this looping be part of the behavior Basic States, we think that the parent state machine (in this case the **Obstacle Task**) should manage whether the basic state should loop or not. This gives us flexibility on when to end tasks (i.e. having timeouts that force state transitions)

cc @Will-Denton @muthuArivoli We'd would like to know your thoughts on these decisions.

muthuArivoli, 2022-11-16:
I really like the interface idea, sounds really cool! A common pattern I've seen is placing the client and server code together, in this case, the interface and the underlying package together. So, the controls_interface would go with the controls package, and task_planning could import it from there. This would allow any changes in controls to be bundled with any changes in the interface and would make sure that the interface still works. Test scripts (like test_state_publisher) could call on the interface instead of directly interfacing with ROS. To package, it would be already included in the way ROS 2 packages python I think, for ROS 1, you would have to add some stuff like catkin_python_setup or something like that. Ultimately though, where it's located is a minor detail, I really like the concept!

For Basic states to be nonblocking, what does that mean? For instance, if I have a state machine A->B->C and B is nonblocking, then once A transitions to B, I would think it would then immediately transition to C. Is that what you mean by nonblocking, or am I not understanding correctly? That feels somewhat confusing, I think transitions should have some sort of condition attached to them. When I think about move_to_local_pose, I would expect it to complete before moving to the next state, not move to the next state even if we're not done moving to the pose. If the issue is that they loop infinitely, then maybe say that they are blocking and are cancellable. Cancellable would mean that the parent state machine could call cancel on the basic state when necessary (for a timeout or some other condition it knows of). But otherwise they would loop forever until they get the desired state. You could also have different states as the end result of cancel, to allow different behavior. Is that what you mean by non-blocking? If you need to have multiple states running at the same time, I think smach has concurrent states or something like that to handle that scenario.

ShaanGondalia, 2022-11-16:
> For Basic states to be nonblocking, what does that mean?

The idea behind nonblocking is that the execution of a state is a single tick instead of a loop. The state machine itself takes feedback from the state (whether thats a completion flag or some other data) to determine whether the state needs to be executed again. This way we can still "loop" a state, it's just that the actual looping is managed by the state machine, not the state itself.

> When I think about move_to_local_pose, I would expect it to complete before moving to the next state, not move to the next state even if we're not done moving to the pose.

In this particular example, the run() method of move_to_local_pose would publish a pose to controls and set a flag that signals whether the robot is currently at the desired location. The state machine that uses this state would then "loop" by repeatedly executing this state until it sees the completed flag. 

Functionally this is the same as keeping the loop inside of the state, but I feel it establishes some design principles that are consistent amongst all tasks. I think it's better to have no tasks execute their own loops instead of some having loops and some simply executing once (I'm imagining servo tasks as an example). 

> If you need to have multiple states running at the same time, I think smach has concurrent states or something like that to handle that scenario.

This is another thing that is simplified with the new approach. Consider the situation where we want state A and B running concurrently, but B requires some input from state A. We can configure a loop with transitions between A --> B --> A on the condition that A is not complete, and then once A is complete we can branch to some other state C. Something similar is possible using the concurrent states, but I'm not sure how smach handles sharing info between concurrent states.

PR #389 (2023-01-14): Refactor how Task Planning uses SMACH
This refactors a large chunk of task_planning. We've extracted interfacing with Controls and CV to "interface" classes which get passed to tasks that need them. Additionally, all of our tasks now extend smach.State directly, instead of an intermediary Task class (which wasn't doing all that much anyway).

Closes #371
closed 2023-01-14 by nickchak21