DukeRobotics/robosub-ros #395: Transition to Compressed Images (DRAFT)

vedarshshah, 2023-01-24:
# Background

All CV code in this repo uses the raw `Image` format when publishing images to topics. This uses a lot of memory, especially for images with large resolutions, and can lead to a slowdown of the entire ROS system. To mitigate this issue, instead of using the raw `Image` format, we need to switch over to the `CompressedImage` format which uses less memory to publish images.

There is two wrinkles in this transition. All bag files that have been recorded in the past use the raw `Image` format. Thus, to play a bag file recorded before this transition and subscribe to the images published by the bag file from our scripts, the scripts must use the raw `Image` format in that case. Therefore, scripts that subscribe to a topic with images must be able to detect whether the images are raw or compressed and handle them accordingly.

The other wrinkle is that, RQT ImageView does not display compressed images unless the topics to which they are published end with `/compressed`. For example, compressed images published to the topic `/camera/front/rgb` cannot be displayed by ImageView unless the topic is changed to `/camera/front/rgb/compressed`.  Therefore, all image topic names will also have to be changed to include `/compressed` at the end so they are compatible with ImageView.

# Purpose

First, transition all topicTransition all CV code that publishes images to a topic to use the `CompressedImage` format instead of the raw `Image` format and add `/compressed` to the end of the topic name. Additionally, enable all CV code that subscribes to topics with images to detect the image type (raw or compressed) and handle it accordingly.

Note that the most challenging part of this transition is not the transition itself, but rather the testing required to ensure all changed code works. Since this transition affects many scripts, each in multiple parts, all affected parts of all scripts must be tested to make sure they work properly. Since each script and part requires different data and methods to test it thoroughly, the task of testing everything will be challenging.

# Tasks
- [ ] Develop a function that converts any given image into any desired format. The function should take two inputs: an image in either the `cv2`, raw `Image`, or `CompressedImage` formats, and the class of the output format desired. It should then perform the conversion. Use the `jpg` compression format. Use the Put this function in `utils.py`. This function will become the backbone of the transition.
- [ ] Transition all publishers in the following files to use the `CompressedImage` format and have `/compressed` at the end of their topic names. For scripts where the topic name is a user-provided argument, do not change it (assume the user will add `/compressed` at the end). Use the function created in the first task to convert images to the `CompressedImage` format. **Transition one file at a time and test each one to make sure it works correctly before proceeding to the next file.**
   - [ ] `depthai_publish_image_stream.py`
   - [ ] `depthai_spatial_detection.py`
   - [ ] `depthai_simulate_detection.py`
   - [ ] `test_images.py`
   - [ ] `usb_camera.py`
- [ ] Enable all subscribers to handle raw and compressed image types.

maxwellmlin, 2023-02-09:
Need to install image-transport-plugins on landside to view compressed images on the cv_gui
```apt install ros-${ROS_DISTRO}-image-transport-plugins```

https://answers.ros.org/question/270881/unable-to-load-plugin-for-transport-image_transportcompressed_sub/

vedarshshah, 2023-02-16:
> I needed to install image-transport-plugins on landside to view compressed images on the cv_gui `apt install ros-${ROS_DISTRO}-image-transport-plugins`
> 
> https://answers.ros.org/question/270881/unable-to-load-plugin-for-transport-image_transportcompressed_sub/

The install has been added to the core Dockerfile.

closed 2023-03-25 by vedarshshah