DukeRobotics/robosub-ros #456: Platform-Independent Joystick Interface

vedarshshah, 2023-09-28:
# Background

Currently, the Thrustmaster joystick must be connected to a Linux computer to use it for controlling the robot. We would like to get rid of the dependency on the Linux computer.

The JavaScript Gamepad API is designed to provide access to joysticks in JavaScript. If this worked with our Thrustmaster joystick, it would make Joystick control cross-platform (Mac, Windows, Linux). If the Gamepad API is accessible in Foxglove, we can develop a Foxglove extension to publish joystick inputs. Otherwise, we can develop a simple website to publish joystick inputs.

Alternatively, we could use Python to interface with the joystick. We could write a set of Python scripts to interface with the joystick and publish the data via Foxglove WebSocket or rosbridge.

# Purpose

Determine if it is possible to obtain joystick data through the methods in the Tasks section. For each method, determine if it works for all, some, or no platforms: Mac, Windows, Linux.

# Tasks
- [ ] JavaScript, Foxglove
- [ ] JavaScript, web browser
- [ ] Python, running directly on personal laptop

AndrewCouncil, 2023-09-29:
pygame works for this, see [here](http://programarcadegames.com/python_examples/f.php?file=move_game_controller.py) for an example to read gamepad input on any device type.
However, I think doing this trough the Foxglove app is best. There is a [Gamepad API](https://developer.mozilla.org/en-US/docs/Web/API/Gamepad_API) that is standard to all browsers (and probably electron? it looks like it is). This API has a lot of npm wrappers, and [react-gamepads](https://github.com/whoisryosuke/react-gamepads) looks promising. Also [joypad.js](https://www.npmjs.com/package/joypad.js) is a javascript library that does similar.

vedarshshah, 2023-10-12:
The [Sideways Teleop](https://github.com/rscova/foxglove-sideways-teleop-extension) Foxglove extension uses React-Gamepad to interface with an Xbox controller and publish Twist messages. We might be able to do the same with the Thrustmaster joystick.

vedarshshah, 2023-10-19:
I have confirmed on my Mac that the Thrustmaster joystick is accessible over JavaScript's Gamepad API. Use [this website](https://greggman.github.io/html5-gamepad-test/) to test joystick compatibility with the Gamepad API.

vedarshshah, 2023-10-19:
I have confirmed that the Thrustmaster joystick is accessible over JavaScript's Gamepad API on Mac, Windows, and Linux.

closed 2023-10-19 by vedarshshah