DukeRobotics/robosub-ros #457: In-depth Software Documentation

vedarshshah, 2023-09-28:
# Background

Each CS team member possesses in-depth knowledge of certain areas of the software. It is vital for this knowledge to be written down, so it can be passed to the next generation of club leaders. This would also serve as a useful resource for new members joining the club to understand various aspects of our software.

# Purpose

Write in-depth documentation on each part of the robot's software. Discuss not just the "what", but also the "why" behind key decisions, tools, frameworks, code, etc.

This is meant to be an ongoing effort through the 2023-24 year.

ShaanGondalia, 2023-09-29:
Would recommend re-recording the software overview video on youtube as well. Its a good resource for people who want a high level understanding of the tech stack and how components work together.

vedarshshah, 2025-02-27:
This is an ongoing process. We've documented some packages in our [wiki](https://github.com/DukeRobotics/wiki). We will continue to work on this, but we don't need a GitHub issue for it.

closed 2025-02-27 by vedarshshah